Lyr3x/Roode #8: Freezing on LOGO

Kabron287, 2019-02-04:
Hello, 
I'm trying to reproduce the project on WEMOS D1 mini Lite with VE53L0/1X and with oled disabled. Compiled OK, but after start i see on serial only Logo and nothing more.
What is the reason?
 __  __       ____
|  \/  |_   _/ ___|  ___ _ __  ___  ___  _ __ ___
| |\/| | | | \___ \ / _ \ `_ \/ __|/ _ \| `__/ __|
| |  | | |_| |___| |  __/ | | \__ \  _  | |  \__ \
|_|  |_|\__, |____/ \___|_| |_|___/\___/|_|  |___/
        |___/                      2.3.1

Lyr3x, 2019-02-04:
Your Radio module cant connect to the gateway. Is your MySensors gateway up and working? Also remove the static `#define MY_NODE_ID 10 // Set a static Node ID if needed` To prevent a NODE_ID override (will remove this soon). 
I never used a wemos d1 mini with MySensors did you checked that the Configuration.h is correct?

Kabron287, 2019-02-04:
Thanks for answer.
I have no radio module. After defining MY_CORE_ONLY, I go a little bit farther. Now I got cyclically:

##### RooDe Presence Detection System #####
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)
wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld

Lyr3x, 2019-02-04:
Is the MySensors init working now? It seems like v0.9.6 does use some libraries which are not supported by the 32-Bit ESP. I ran into the same issue yesterday. 
I am working at a huge refactor of the whole architecture to support the ESP and MQTT.
Checking https://www.mysensors.org/build/connect_radio it seems you need to connect a extra radio for the wemos running as mysensors node. Thats why i am building MQTT support for the ESP's. If you have an NRF24L01+ and an Arduino ready everything will work oob.
In a few days you may check the v0.9.6-rework branch to use the wemos d1 mini with MQTT instead of MySensors.

Kabron287, 2019-02-04:
I definitelly do not need any radio except the ESP8266 internal, as I use Blynk.
What should I setup to exclude as much as possible from MySensors support to leave clean People Counter?
Now I tested with 0.9.5

Lyr3x, 2019-02-04:
Ah alright. For now there is no Blynk support available. For now there is no easy code change to support blynk available. 
You need to wait until i finished the refactor and then you can add a new blynk transmitter.

Kabron287, 2019-02-04:
Also they have a very interesting solution with the single VL53L1X sensor:
https://community.st.com/s/question/0D50X0000A7VWoMSQW/is-vl53l1x-people-counting-source-code-available

A subject to think

Lyr3x, 2019-02-05:
@Kabron287 it's a little more complicated then this. I checked their code a few month ago and they do nearly the same but with a few drawbacks where some edge case scenarios will fail. Their code is just a demo code. The core logic is identical but tuned my code a lot in the last two years to make it a lot more reliable for edge cases.

I'll implement the VL53L1X soon to make it possible to use one sensor and the SPAD array.

If you like to test roode you can now checkout the v0.9.6-rework branch and setup mqtt, for example with mosquitto.

Lyr3x, 2019-02-05:
If you like you can start to implement a new Transmitter type (Blynk) based on v0.9.6-rework.
MySensors for Arduino boards and MQTT for ESP boards is working (not ready yet).

Lyr3x, 2019-02-12:
Closing here now. Feel free to contribute with Blynk support. I'll add Vl53L1X support in the next few days. I'll integrate my test code in the project soon

closed 2019-02-12 by Lyr3x