Lyr3x/Roode #19: FYI: made an esphome version of this

diplix, 2020-05-09:
based on your work i managed to convert this (the 1.0 version) into a working [esphome](https://esphome.io) configuration: https://github.com/diplix/esphome-configuration

i’m using two of these and they are working quite reliably fro some weeks now. one esp8266 tof-sensor is combined with a door sensor, OTA updates are working  (thanks to esphome) and on the fly calibartion should be doable as well.

Lyr3x, 2020-05-09:
Awesome work! Due to my master Thesis I have no time at all at the moment... I would love too see and check your work! 

Are you interested in open a PR which includes the esphome stuff?

diplix, 2020-05-09:
iff you want to test my configuration you could (after installing esphome and checking/configuring the platform in `tof_sensor.yaml`) just run `esphome tof_sensor.yaml compile` to see if it compiles and `esphome tof_sensor.yaml run` to compile and upload on a esp8266 or esp32.

closed 2020-10-15 by Lyr3x