Lyr3x/Roode #20: Maybe use a new people counting algorithm

leyuansugar, 2020-10-09:
**Is your feature request related to a problem? Please describe.**
In this version, the counting algorithm is derived from gesture direction recognition algorithm. In some practical situations, it will be inaccurate. For example, if a person walks from the outside to the room, stays near the sensor, does not enter the room and then turns out, the sensor will mistakenly believe that someone has entered the room.

**Describe the solution you'd like**
Use st's people counting algorithm.

**Describe alternatives you've considered**
My coding is very bad.I tried to convert ST's stm32 people counting code to esp32 code,but I failed.
Maybe you can try this.

**Additional context**
FYI
St's source code: [https://www.st.com/en/embedded-software/stsw-img010.html](https://www.st.com/en/embedded-software/stsw-img010.html)
stm32 to esp32 example:[https://github.com/kazkojima/esp32-vl53l1x-test](https://github.com/kazkojima/esp32-vl53l1x-test)

Lyr3x, 2020-10-15:
The code already handle the situation that you partly go into the sensor and go back in the other direction. Interesting that ST themselves published the adapted direction detection! Will look into it, if there is anything special that might help. Thanks for the heads up!

leyuansugar, 2020-10-15:
Yes. I check the code and test the situation,but the results were not very satisfactory. When I go into the sensor(into detection area), and go back,the num of people will add.Every time I do this test,it will add 1 people num. I already test the st's algorithm with vl53l1x and esp32,it's worked fine. If there is any help,I can sent the arduino code of this to you.Maybe you can add to your roode project. I think the esp-idf is better than arduino-ide,but my coding skill is bad.Maybe you can do this.

markjones89, 2020-10-27:
For interest here is a very similar project using the ST code https://github.com/Andrea-Fox/peopleCounter

Lyr3x, 2020-10-27:
Very interesting too see more projects coming up :)

I will try to implement the latest ST code this weekend. I can't reproduce your behavior as my setup is working nearly flawless for me.

Smexah, 2020-12-02:
Hi, 

First of all: Thanks for the code! I love the idea of this project. 
I have implemented your code on an ESP like 3 month ago. After a little time of reviewing what you did there - all worked fine. Little more documentation would be nice, but it was ok anyway. So, I am "testing" since then.  

First, I mounted the TOF in the doorframe (1.95m), facing down the frame. In many cases, it worked fine. Biggest problem is the "over counting" so that I sometime have up to 15 imaginary people around me ;) 
Then I remounted the sensor over a door (2m) with the door opening to the other side. But in my opinion it made things worse. 
For example: when you leaving the room by opening the door and closing it in the same move, the counter do not decrease. And if you are entering again, you have a new imaginary friend. 

After encountering the “door issue” I mounted it over a frame with no door in it. I would say it works better, but after all not perfect. I have tried different angles and positions on the frame, but nothing seems to be optimal. 

Ideas as feedback: A little more doc, like ports and maybe a “how to install” with steps. For better understanding, an image or schema of your real world hardware setup/installation (like sensor position) would be a wonderful extension.
Anyway thanks for your work.
Have a nice day :)

Lyr3x, 2021-08-10:
I will start working on Roode again. And check the latest ST code. However i cant reproduce the behavior you describe @leyuansugar The ST code does not something special detection wise. However it is indeed a simpler implementation

leyuansugar, 2021-08-10:
@Lyr3x Just go into the sensor(into detection area), and go back，do not leave the testing area during the entire process。

Lyr3x, 2021-08-10:
I know what you mean because I worked a lot on that particular issue, and it is working fine for my setup. But I know that the positioning as well as surfaces makes a huge difference in accuracy. nevertheless, I will take a look into the ST lib and will switch my own MQTT implementation to esphome.

If you like to support, I ham happy to accept PRs

Sent with a Spark

Lyr3x, 2021-08-10:
I'll update the issue if I have something to test :) thank you for your hint!

leyuansugar, 2021-08-10:
Ok, ST updated the code for the headcount some time ago, you can see it here https://www.st.com/en/embedded-software/stsw-img010.html My suggestion is to migrate the algorithm for this code to esp32, I tried it and it worked well, but that was the first version and not this latest version of ST.

Lyr3x, 2021-08-17:
Getting familiar with esphome atm and added a draft branch with esphome config and an improved algorithm. Thanks to @diplix who provided me his changes. 

I added the calibration again and removed old code. I will test that branch and fix stuff over the next days and then start to rework the algorithm. Feel free to submit PRs 👍

aerodolphin, 2021-08-18:
That sensor works well but with too much light with the same spectrum (sun) will miss counting. Can you test that case?

Lyr3x, 2021-08-19:
Makes sense, that's a sensors limitation which we can mitigate with a well designed housing. At least to a certain degree. Also you can change the Position of the Sensor. Where do you installed it?

Lyr3x, 2021-08-20:
@leyuansugar Check the v1.1 branch. Would appreciate feedback

leyuansugar, 2021-08-20:
@Lyr3x In long distance mode, the parameter setTimingBudgetInMs can be changed to 33 to get better performance. also in practice, different roi area settings can bring very different results, or you need to test the suitable roi area settings several times for different environments. i'm very sorry i don't have time to do some tests lately, i can only give you some of my experience from testing at that time.

Lyr3x, 2021-08-20:
@leyuansugar Thanks for the hint. The old algorithm used 33ms (we can even go down to 22 without issues in short mode) but I missed to adapt that accordingly.

Lyr3x, 2021-08-20:
Adapted all pieces to use the correct timings. Indeed it improved the latest version. Ofcours everyone needs to set up the personal ROI. That is nothing I can generally provide. The standard settings should work in most cases, like chest height wall to wall mount with ~1m distance

closed 2021-08-29 by Lyr3x