Lyr3x/Roode #27: Random disconnects of the ESPHome API

Lyr3x, 2021-08-20:
**Describe the bug**
When controlling Roode from HA it come sometimes to an disconnect after a command is sent

**To Reproduce**
Steps to reproduce the behavior:
Sent recalibrate command

**Expected behavior**
Recalibration occurs, but after the recalibration started the esp disconnects for a few seconds

Lyr3x, 2021-08-20:
We can't enable both mqtt and the native api without stability issues. We need MQTT to receive commands and react on them in the code. Currently the native api is not capable of that. However, it seems that they develop that possibility.

Lyr3x, 2021-08-29:
Cleaning up exposed sensors + setting a higher Timeout mitigated the issue well

noxhirsch, 2021-09-14:
> We can't enable both mqtt and the native api without stability issues. We need MQTT to receive commands and react on them in the code. Currently the native api is not capable of that. However, it seems that they develop that possibility.

@Lyr3x Sorry to respond to a closed issue here, but shouldn't this be possible with user-defined services so that MQTT isn't needed anymore?
https://esphome.io/components/api.html#user-defined-services

I use these user-defined services to trigger specific commands or send position data to my blinds:
[image]

Lyr3x, 2021-09-14:
Thank you very much! I didn't know about that. I'll reopen the issue and will test that to ditch MQTT completely.

Lyr3x, 2021-09-14:
Do you know if it's possible to use just a lambda in the `then` part?

noxhirsch, 2021-09-14:
If that's what you mean, then yes, it's possible:

```
    - service: slow_position
      variables:
        pos: float
      then:
        - [...]
        - lambda: 'id(tilt).update();'
        - [...]
```
(this is snippet of my blinds-script)

Lyr3x, 2021-09-14:
Awesome! Thanks again

Lyr3x, 2021-09-23:
With #43 I removed MQTT and switched to NativeAPI all the way. Services are registered in the example configurations to achieve resetting, counter setting and recalibration.

closed 2021-09-23 by Lyr3x