Lyr3x/Roode #28: Use of external_components

DutchDeffy, 2021-08-23:
**Is your feature request related to a problem? Please describe.**
It takes quite some configuring and moving files to get started with this. The advantage of esphome can be exploited more, I think.

**Describe the solution you'd like**
Instead of moving all the files over and stuff. Use [external components](https://esphome.io/components/external_components.html?highlight=external ).

```
external_components:
  - source: github://Lyr3x/Roode
  ```

Lyr3x, 2021-08-23:
You might totally right! I just started working with ESPHome. I am sure there is a lot of room for improvements. Thank you for the specific hint. I will change that. Feels free to open PR's if you know fixes for low hanging fruits

DutchDeffy, 2021-08-23:
Nice! I saw you already looked into the STMicroelectronics and Andrea-Fox's code? Can you tell me is there any difference in the way it detects people? Any different approach of the algorithm?

Lyr3x, 2021-08-23:
I made some minor adjustments. Like using the latest code base from ST and improved reliability in a way that I introduced the checkDataReady() functions and using larger ROIs (8x16 instead of 8x8 or 4x4). However I am still optimizing the code. 
I maybe also switching back to the official API instead of using the spark fun lib. 

If you test: use the v1.1 branch.

Lyr3x, 2021-08-23:
@DutchDeffy How do you flash your esp? I checked the external_component thing and don't think that would really help? I would need to move the codes to a separated components folder. 
What I am doing is the following:
1. git clone <repo>
2. Modify secrets.yaml etc.
3. esphome run peopleCounter.yaml

If I move the code to a "component" do literally the same. Am I wrong?

DutchDeffy, 2021-08-23:
You don’t have to clone the repository. Once the external component is set up, I think. You can have a look at Zuidwijk’s dsmr component. https://github.com/zuidwijk/dsmr/tree/b42d4488c5beb2d3c8c3351b3d19a6e4a03d5540 
Which got natively supported in Esphome version 2021.8.0. But before was build as external component. 

Once an external component is set, you only need to copy all the yaml config files (with common if used) to flash the esp from the Esphome dashboard.

Lyr3x, 2021-08-23:
Ah esphome Dashboard is the magic word. I understand that but I don't use the dashboard. Nevertheless I can try to restructure the repo to make that work 🙂

DutchDeffy, 2021-08-23:
Without the dashboard and using platform io it is also possible to install software to your esp, as you said with esphome run. The getting started explains it very well. https://esphome.io/guides/getting_started_command_line.html
The dashboard is much more easy to use, if you ask me and together with home assistant (supervised) it is very powerful.

Lyr3x, 2021-08-23:
Yeah I know the dashboard but I am a CLI guy :) totally understand why people use it. 

I will work on that!

Lyr3x, 2021-09-13:
This is in progress now. It will definitely part of the next release. 
@DutchDeffy thanks for the hint 👌

Lyr3x, 2021-09-23:
Part of #43

closed 2021-09-23 by Lyr3x