Lyr3x/Roode #30: Natural light interference

jcastro, 2021-08-23:
Hello!
Thanks so much for sharing this. I've been long testing other repos who tried to do same counting with this sensor but in all of these I've found the sensor is not reliable at certain wavelengths during day. Tried several different sensor manufacturers, cases, position in the door frame and so on, always with same result.

I've installed today this repo (it's night time for me now) with the hope that somehow it will work better and software will more or less compensate for the hardware issues this sensor has, so will report back!

In any case, I wanted to also ask you if you have ever considered using instead the thermal sensor AMG8833 for doing the counting, which it seems what they're using here https://www.hiome.com/. With this sensor it seems we won't have at least the natural light problem. There's already some repos trying to come up with person detection (not counting) but it could be useful for inspiration if you ever consider it https://github.com/TheRealWaldo/esp8266-amg8833

Thanks!

Lyr3x, 2021-08-24:
Hey!
yes ToF Sensors which are emitting 940nm waves are naturally affected by direct sunlight. In the last years, i barely had issues with that due to the fact that i do not face the sensor to direct sunlight. The best possible position, where sunlight does play a minimal role is the ceiling mounting. 

A way to mitigate the issue (currently not available anymore) is to add a light sensor and recalibrate (thats already working) to a certain delta. It is not in our interest to get precise distances, however the relation between we enter a zone and leave a zone needs to be consistent. 

I did not know the AM8833, and it sounds very interesting. I tried to achieve counting with other thermal sensors before, and it was horribly unstable. The sensors I used were extremely sensitive to temp changes. 
I leave that issue open as I surely look into it.

jcastro, 2021-08-24:
Thanks for your reply @Lyr3x. Based in my experience it doesn't even have to be in direct sunlight, in fact mine is not!. I've tried setting it up in the side of the door, on the top and also on the ceiling, in all cases at some point in time in the day they will act up weird. Even if I leave it on my desk, which never gets directly sunlight will do the same.

Hopefully there's some fix or help with other sensor as you said, or maybe ST releases other newer sensor, or other brand. I'm guessing there's light sensors, or some kind of sensor that can work reliably with sunlight environment, maybe 2 of those could make it as well.

I'm sorry to bother, just my two cents. I've been wanting to have some kind of occupancy detection for soooo long that I'm super excited about your project. Thank you!

Lyr3x, 2021-08-24:
Just to collect information. May you post links to the sensors you tried?

jcastro, 2021-08-24:
> Just to collect information. May you post links to the sensors you tried? Another thing the sensor is kind of sensitive: temperature changes (+-8°) makes a huge difference. The code does a temp calibration on boot already. I will expose that via MQTT as well so you could recalibrate that too.

Absolutely!
- https://www.waveshare.com/vl53l1x-distance-sensor.htm
- https://es.aliexpress.com/item/4000074204979.html?spm=a2g0s.9042311.0.0.274263c0KanD2K
- https://es.aliexpress.com/item/1005001953913055.html?spm=a2g0s.9042311.0.0.274263c0KanD2K
- This one but with a black cover on the top of the sensor (https://es.aliexpress.com/item/1005001966960395.html?spm=a2g0s.9042311.0.0.274263c0KanD2K)

Yes, there might be those temperature variations in my room on summer time, as I leave my windows open at night. It's probably not happening for winter tho, the temperature varies but not that much

jcastro, 2021-08-24:
Here's also a pastebin from the log of the sensor https://pastebin.com/AksUYgX7 (Waveshare one, which is supposedly a "good brand"

It's sitting on my desktop looking up to the ceiling and it have counted 2 people (not present on this log as this was before). Here never gets direct sunlight either

Hope it helps!

[image]

Lyr3x, 2021-08-24:
Are these values already wrong? They look good to me

jcastro, 2021-08-24:
> Are these values already wrong? They look good to me
No, these are more or less okay, from desk to ceiling there should be more or less that distance. What I meant to say that at some point in the day the natural light will interfere with the sensor and make it go nuts

jcastro, 2021-08-30:
Just as an update to this issue in case anyone else sees it. It looks like that having a case for the sensor where it sits a bit recessed reduces or eliminates this light interference. I'm still testing so will post more when I know

Also, super important: USE A GOOD BRAND for the sensor. I've got 2 cheap chinese sensors and they work very bad, while others like Waveshare offer rock solid and stable measurements

Lyr3x, 2021-09-03:
Will close this issue, as we can not due much about the interference.

closed 2021-09-03 by Lyr3x