Lyr3x/Roode #36: Rework ROI detection

Lyr3x, 2021-09-13:
The ROI detection needs to be reworked. It should find the best ROI settings defined by distance and target surface size.

Lyr3x, 2021-10-06:
I need to test this thoroughly. I am not sure right now if this will improve things or make them worse. From my latest findings I think the ROI size itself does not matter as much as the target surface material and sensor alignment.

Lyr3x, 2021-11-17:
I implemented a fairly simple change to produce one of the following ROIs
- 4x8
- 6x12
- 8x16

I am not sure right now if that helps in all scenarios and I need to do more testing. Feedback is much appreciated!

Lyr3x, 2021-12-03:
ROI center settings must be reworked too as there are three different situations where the ROI center points need to be set. Currently the center is not working properly everywhere

CarsonF, 2022-01-01:
New schema suggestion
```python
cv.Optional("roi", default={"height": 16, "width": 6}): cv.Any(
    cv.one_of("auto"),  # formerly "roi calibration"
    cv.Schema({
        cv.Optional("height", default=16): cv.int_range(min=4, max=16),
        cv.Optional("width", default=6): cv.int_range(min=4, max=16),
    })
),
```

Maybe default switches to `auto` once the implementation is deemed better