Lyr3x/Roode #47: Hardware recommendations

andreimoraru, 2021-10-08:
**Is your feature request related to a problem? Please describe.**
I am interested to build the Room Detection solution and I am looking for some reliable hardware recommendations. The motivation to have reliable hardware is to avoid unexpected behavior down the road due to the fact that I have saved a couple of dollars on hardware.

**Describe the solution you'd like**
I would like to have a list of recommended hardware that proven to be reliable.
So far I have compiled the following hardware list but not sure which one to choose.
ESP32 boards:

- esp32-s3-devkitc-1-n8r8 - pretty modern development boards and decent prices (around 15USD or 12 EUR or 13GBP at different shops - digikey.com, mouser.com, tme.eu etc.). As alternative, one can choose esp32-s3-devkitc-1-n8r2 or esp32-s3-devkitc-1-n8 - they are more often in stock at reputable retailers.
- Wemos D32 Pro from wemos.cc
- HUZAHH32 https://www.adafruit.com/product/3405 . It costs about 20USD
- ESP32 Thing Plus from Sparkfun https://www.sparkfun.com/products/15663 . It costs about 22USD.

VL53L1X sensors:
- from waveshare.com https://www.waveshare.com/product/modules/vl53l1x-distance-sensor.htm . It costs about 15 USD
- from Sparkfun https://www.sparkfun.com/products/14722 . It cost about 22USD
- from Adafruit https://www.adafruit.com/product/3967 . It costs about 15USD
- from pololu https://www.pololu.com/product/3415 . It costs about 12 USD

**Describe alternatives you've considered**
I studied the open and closed issues for this git project and so far found only a mention from a user reporting that waveshare VL53L1X sensor proven to be much more reliable than cheaper versions bought from some other stores.

**Additional context**
I plan to buy hardware for room detenction for 5 doors in my house initially, hopefully later will extend to all other rooms.

Lyr3x, 2021-10-11:
These are very good and also tough questions. We had a lot of discussions in our discord server about various breakout boards and behaviors. I can’t give you an exact answer as it seems to depend also a lot on your own environment. What I can assure is that brands like Wemos, SparkFun and Pololu deliver high quality products. I personally use wemos Esps as well as some clones and stick to the pololu vl53l1x breakout board which is working flawless. 

I also can tell you that you should avoid cheap chinese products by any means.

Lyr3x, 2021-11-30:
I am in the process of defining supported hardware and evaluate some stuff before writing it down. But that being said it Wemos/Lolin ESPs and Pololu sensors are in the support list

Lyr3x, 2021-12-03:
Added a section to the README in #55

closed 2021-12-03 by Lyr3x