Lyr3x/Roode #49: ESP8266 WiFi not connecting

grego1981, 2021-10-15:
**Describe the bug**
Used the latest yaml example, the firmware is compiled just fine, I upload it without an issue and restart the module. On restart if the sensor is not connected the WiFi is connected properly after attempting to find the sensor. If the sensor is connected Wifi fails to connect.

**To Reproduce**
Connect the sensor to SDA & SCL

**Expected behavior**
The ESP8266 should connect to WiFi with sensor connected

**Additional context**
Attached 2 log files one with sensor online and one offline.
[with_sensor.txt](https://github.com/Lyr3x/Roode/files/7351916/with_sensor.txt)
[without_sensor.txt](https://github.com/Lyr3x/Roode/files/7351921/without_sensor.txt)

Lyr3x, 2021-10-15:
Thanks for reporting this. 
What power supply do you use?

Lyr3x, 2021-10-16:
Can you share your exact hardware information? 
e.g.:
ESP model: Wemos D1 Mini
VL53L1X: Pololu
Wiring: All pins

grego1981, 2021-10-16:
ESP model Wemos with ESP8266EX chip
[image]
[image]
VL53L1X form Aliexpress ([the black one](https://www.aliexpress.com/item/1005001934379232.html?spm=a2g0s.9042311.0.0.27424c4d8PTfN3))
D1 --> SCL
D2 --> SDA
3V3 --> VIN used 3v3 & GND both next to the D1D2 pins as well the one of the opposite side.
GND --> GND

grego1981, 2021-10-16:
And this is my ESPHome config file:

substitutions:
  devicename: bathprsensor
  friendly_name: $devicename

external_components:
  refresh: always
  source:
    type: git
    url: https://github.com/Lyr3x/Roode
    ref: master

esphome:
  name: $devicename
  platform: ESP8266
  board: d1_mini
  libraries:
    - pololu/VL53L1X@^1.3.0

wifi:
  networks:
    - ssid: !secret ssid1
      password: !secret ssid1_password
  use_address: $devicename
  fast_connect: True
  power_save_mode: none
  domain: .local

api:
  password: admin
  reboot_timeout: 60min
  services:
    - service: set_counter
      variables:
        newCount: int
      then:
        - lambda: "id(roode_platform)->sendCounter(newCount);"
    - service: reset_counter
      then:
        - lambda: "id(roode_platform)->sendCounter(0);"
    - service: recalibrate
      then:
        - lambda: "id(roode_platform)->recalibration();"

ota:
  password: admin

web_server:
  port: 80
  auth:
    username: admin
    password: admin

# Enable logging
logger:
  level: DEBUG

i2c:
  sda: 4
  scl: 5

roode:
  id: roode_platform
  address: 0x29
  update_interval: 100ms
  calibration: true
  roi_height: 16
  roi_width: 6
  max_threshold_percentage: 80
  roi_calibration: false
  invert_direction: true
  restore_values: true

switch:
  - platform: restart
    name: $friendly_name Restart
binary_sensor:
  - platform: status
    name: $friendly_name Status
  - platform: roode
    presence_sensor:
      name: $friendly_name presence

sensor:
  - platform: roode
    id: hallway
    people_counter_sensor:
      id: peopleCounter
      name: $friendly_name people counter
    distance_sensor:
      name: $friendly_name distance
      filters:
        - delta: 10.0
    max_threshold_zone0:
      name: $friendly_name max zone 0
    max_threshold_zone1:
      name: $friendly_name max zone 1
    min_threshold_zone0:
      name: $friendly_name min zone 0
    min_threshold_zone1:
      name: $friendly_name min zone 1
    roi_height:
      name: $friendly_name ROI height
    roi_width:
      name: $friendly_name ROI width

  - platform: wifi_signal
    name: $friendly_name RSSI
    update_interval: 60s

  - platform: uptime
    name: Uptime Sensor
    id: uptime_sensor
    update_interval: 60s
    internal: true
    on_raw_value:
      then:
        - text_sensor.template.publish:
            id: uptime_human
            state: !lambda |-
              int seconds = round(id(uptime_sensor).raw_state);
              int days = seconds / (24 * 3600);
              seconds = seconds % (24 * 3600);
              int hours = seconds / 3600;
              seconds = seconds % 3600;
              int minutes = seconds /  60;
              seconds = seconds % 60;
              return (
                (days ? String(days) + "d " : "") +
                (hours ? String(hours) + "h " : "") +
                (minutes ? String(minutes) + "m " : "") +
                (String(seconds) + "s")
              ).c_str();
text_sensor:
  - platform: roode
    version:
      name: $friendly_name version
  - platform: roode
    entry_exit_event:
      name: $friendly_name last direction
  - platform: template
    name: $friendly_name Uptime Human Readable
    id: uptime_human
    icon: mdi:clock-start

Lyr3x, 2021-10-30:
I think that it has nothing do do with my code, but with esphome itself. I am not available till tomorrow evening. May you join the discord and well built Test code and debug this?

We can post the results here afterwards.

grego1981, 2021-11-01:
No problem I can do it, just tell me when!

Lyr3x, 2021-11-07:
https://discord.gg/RK3KJeSy

DutchDeffy, 2021-11-10:
I had the same problem with my esp8266. This was the error in the logs.

`: Event: Disconnected ssid='***' bssid=B0:**:**:**:FB:** reason='Auth Expired'
`
 Removing the lines starting from use_address until domain solved the issue for me. So this

```
wifi:
  networks:
    - ssid: !secret wifi_ssid
      password: !secret wifi_password
  use_address: $devicename
  fast_connect: True
  power_save_mode: none
  domain: .local
```
Will be this:
```
wifi:
  networks:
    - ssid: !secret wifi_ssid
      password: !secret wifi_password
```

Lyr3x, 2021-11-11:
Could it be that fast_connect is the issue here? Can you check if you enable all other options again but leave fast_connect disabled?

DutchDeffy, 2021-11-11:
I think it's an issue of my wifi router. It's working fine now as is.

Lyr3x, 2021-11-11:
So it has to do with fast connect ?

Lyr3x, 2021-11-23:
Will close the issue as it is most likely not an issue with the RooDe code.

closed 2021-11-23 by Lyr3x