Lyr3x/Roode #56: Weird bug with missing esphome/components/switch.h

jseidl, 2021-12-01:
**Describe the bug**
Just installed the external_component on ESPHome to try it out, first build I get:

```
fatal error: esphome/components/switch/switch.h: No such file or directory
```

Which is weird since this include seems perfectly valid according to ESPHome's documentation. I'm still unsure if it's something local to me or something else but I've cleaned my whole ESPHome directory (left only the YAML files) and this was still happening. I've searched the code and didn't see Roode using `switch` stuff, so I just commented the include and everything worked perfectly. Maybe a leftover?

**To Reproduce**
Steps to reproduce the behavior:

Include Roode via the following snippet:

```
external_components:
  - source: github://Lyr3x/Roode@master
    components: [ roode ]
```

Try to build your configuration. Build fails with

> fatal error: esphome/components/switch/switch.h: No such file or directory

**Expected behavior**

Clean build.

**Screenshots**
N/A

**Additional context**
This might be a leftover include. Still unsure why it happens, glad removing it results in no issues so far.

Lyr3x, 2021-12-01:
Can you please post your complete config? Remove any secrets upfront. Good thing is I can't reproduce it with a fresh build right now.

Lyr3x, 2021-12-01:
Will be removed in #55. Thanks again!

closed 2021-12-01 by Lyr3x