Lyr3x/Roode #63: Just tested v1.3.3

diplix, 2021-12-04:
i’ll just note what stuck out while testing it. not everything i conclude is meant as a feature request, it’s more a field report.

first off, my VL53L1X breakout board has an `xhut` pin, which needs to be set HIGH to work. i connected it to pin 23 and added the folowing yaml:
```
esphome:
  # ...
  on_boot:
    then:
      - output.turn_on: xshut

output:
  - platform: gpio
    pin: 23
    id: xshut
```
during my first tests, because of the missing xshut, roode "Failed to detect and initialize" the sensor and shut down because of a loop.
 https://github.com/Lyr3x/Roode/blob/d946b95c98a8048f7b0cb008d393a8c1ef50931a/components/roode/roode.cpp#L42

that’s kind of a pain, because it renders the esp32 useless and unable to over the air updates. logging the error and letting esphome do it’s thing would be much nicer.

next problem i had was the watchdog. as long as the API was configured calling calibration caused a watchdog induced reboot. disabling initial calibration

```
roode:
  # ...
  calibration: false
```

enabled esphome to run and loop, however there were constant API disconnections (every 1-2 seconds) . increasing the  `update_interval` to ~ 200ms or 400ms helped a little, but the API still disconnected eveery few seconds.

removing  the API
```
#api:
#  password: !secret api_password
#  reboot_timeout: 60min
# ...
```

and unsing `mqtt:`helped: initial calibration ran through and an `update_interval ` of 10ms was ok.

to substitute the API services, i configured mqtt subscribes:

```
mqtt:
  broker: !secret mqtt_broker
  username: !secret mqtt_username
  password: !secret mqtt_password
  port: !secret mqtt_port
  reboot_timeout: 20min
  
  on_message:
    - topic: ${devicename}/services/set_counter
      qos: 0
      then:
        - lambda: "id(roode_platform)->sendCounter(atoi(x.c_str()));"

    - topic: ${devicename}/services/reset_counter
      qos: 0
      then:
        - lambda: "id(roode_platform)->sendCounter(0);"

    - topic: ${devicename}/services/recalibrate
      qos: 0
      then:
        - lambda: "id(roode_platform)->recalibration();"
```

however calling `id(roode_platform)->recalibration();` while roode is runing still calls the watchdog. setting and resetting the counter works fine this way.

since esphome 2021.11 there is a feature called `entity_category` ([see esphome docs](https://esphome.io/components/sensor/index.html)). it would be nice to make the following `platform: roode` sensors "diagnostic": `distance_sensor`, `max_threshold_zone0`,`max_threshold_zone1`,`min_threshold_zone0`,`min_threshold_zone1`,`roi_height`, `roi_width` and `{"text_sensor":["platform":"roode", "version":[]]}`

other than that roode works realy nice and reliably. i like how the auto-configuration works and the many diagnotic sensors for debugging and testing.

here are links to the parts i used:
* https://de.aliexpress.com/item/32957084608.html
* https://de.aliexpress.com/item/1005003056543472.html

Lyr3x, 2021-12-04:
1. All my VL53L1X Sensors (see hw list) do not use the XSHUT but you already found out how to get that working!
2. On initial boot it might happen that you get a reboot during the calibration but after a second attempt it should work without issues. I just stumbled over wdt issues with increases timing_budgets... But maybe your ESP uses a different default wdt ? Because on my Wemos D1 Minis this issues is not reproducible.  I can try to add more `yield()` calls in the process to give the ESP more time doing its things.
3. Thanks for the entity_category hint! I will implement that in the upcoming version!

Lyr3x, 2021-12-04:
Added the `ENTITY_CATEGORY_DIAGNOSTIC` for all sensors in #55

Lyr3x, 2021-12-04:
> during my first tests, because of the missing xshut, roode "Failed to detect and initialize" the sensor and shut down because of a loop.

The issue here is you will never get your Sensor up and running afterward because it will never jump back to the setup function. I added this to prevents error searching where the error lies in the sensor communication itself. Does that make sense ?

Lyr3x, 2021-12-04:
I removed the while blocker in #55. If this results in confusion for users i can add it back.

Lyr3x, 2021-12-04:
> first off, my VL53L1X breakout board has an xhut pin, which needs to be set HIGH to work. i connected it to pin 23 and added the folowing yaml:

All VL53L1X boards have an XSHUT pin which is automatically pulled to HIGH but you can put it to LOW to shut down the board. If you dont use the pin at all your sensor will work OOB

closed 2021-12-18 by Lyr3x