Lyr3x/Roode #66: Provide Recalibration as button

Lyr3x, 2021-12-11:
As of 2021.12 ESPHome and HA support button entities which do a certain action on press. That is something that were missing to implement a proper recalibration, our counter reset button.

Lyr3x, 2021-12-22:
```
- platform: template
    name: $friendly_name Recalibrate
    on_press:
      - lambda: id(roode_platform)->recalibration();
```

This is working for now but it should be a `-platform: roode` button

sunshine-hass, 2022-01-03:
Hello, have you actually tested this project?
I tested https://github.com/Andrea-Fox/peopleCounter before, and it can meet my needs. Now the home is being renovated and I need to integrate the ESPhome version.

Lyr3x, 2022-01-03:
I don't get your question. Andreas Version is a very basic Version and not really comparable anymore.

closed 2022-01-15 by Lyr3x