Lyr3x/Roode #72: Continuous distance readings

ruimarinho, 2021-12-22:
Hi @Lyr3x,

It seems like I've been able to enable just the `distance_sensor` in order to get continuous readings by disabling the people counter sensor.

Is this a supported use-case? I see that the distance changes a lot even if the sensor is completely stable, which might indicate the distance mode is not correctly configured. Should I be manually configuring the sensor to mode `2` (in my case)?

Thanks,

Lyr3x, 2021-12-22:
Sure, it's possible to only use the readings. Remember that the calibration already sets a mode and timing budgets,which are fairly tight as we don't need super stable readings but rather quick one. Use the manual mode and set your own budget and you should be good to go!

Lyr3x, 2021-12-22:
#73 Will fix the manual mode and you should configure something like this:
```
roode:
  id: roode_platform
  i2c_address: 0x29
  update_interval: 10ms
  manual:
    sensor_mode: 3
    roi_height: 16
    roi_width: 6
    manual_threshold: 1280 
    timing_budget: 50
  restore_values: false
```

ruimarinho, 2021-12-23:
That looks good! What if I'm simply interested on distance readings (e.g. for detecting a parked car, etc) and not human presence?

Lyr3x, 2021-12-24:
You can just remove the people counter and use the manual mode which gives you distance and you could also use the presence sensor to see if the area is occupied. However you will have two zones, which might not be a problem

Lyr3x, 2021-12-28:
Ill close this, feel free to join our discord server for more questions :)

closed 2021-12-28 by Lyr3x