Lyr3x/Roode #81: Battery operated

electricfire, 2022-01-01:
Hi will this solution work on a battery i am using Home Assistant with ESPHome, if so what type of a battery and also how long do you think it will last.  If not...  as i am using the D1 mini i wonder if this can be put into deep sleep and use the count trigger to wake up the D1 mini to send the data.  Maybe use a ATTiny for the interrupt (i am no expert i have seen someone else use this approach).  But i guess the question there is how to trigger the  ATtiny from the VL53L1X as the people counting is software.  Let me know if this is even a possibility.

Lyr3x, 2022-01-03:
How about try it out and tell us? You could add a simple PIR used as an interrupt and configure that in esphome. It's hard to tell how long a battery will last then.

redmusicxd, 2022-01-03:
The VL53L1X can trigger an interrupt on the GPIO1 but that would first require a bit of  modification of this software to configure the threshold at which the sensor will trigger also ESPhome has a deep sleep prevention thing to do OTA but I guess the problem would be that it has to reconnect to Wi-Fi when it wakes up, hoping someone can pull this one day.

Lyr3x, 2022-01-03:
It will not help to use the VL53L1X as an interrupt device because it will wake the esp every few ms.

electricfire, 2022-01-03:
I guess we would have to have an external sensor to wake up the VL53L1X, because if you have two people enter the room, i am guessing using the interrupt on the VL53L1X would just notify that something happened as opposed to two people have just been identified unless i have missed something here.  FYI i am no expert in this field i am just curious with this whole automation idea so i will try and give it a go, but there are way more smarter people out there, just wondered if this was thought about as in running it on a battery

Lyr3x, 2022-01-03:
As said before we need e.g a PIR as interrupt. Done that years back  when roode was in a different state. Just attach a PIR and configure the deep sleep and you can start to experiment with battery powered settings.

redmusicxd, 2022-01-03:
Oh yeah, you’re right as the sensor will just stay LOW as long as it’s within the threshold , don’t mind me just a guy with ideas :)
Though the PIR would be the closest solution as there are sensors like Xiaomi’s that you can modify their blind time down to 5 seconds and reportedly the battery doesn’t change(didn’t tried yet so don’t take my word for it) but those afaik don’t have an interrupt

Lyr3x, 2022-01-03:
You would use something like an HC-SR501 for that purpose

electricfire, 2022-01-03:
@Lyr3x ok so it's been looked into and by the sounds of it, it's not a feasible solution, i guess if we can get a year out then it would be worth, but probably not worth pursuing as i guess even trying to add more batteries it would come down to traffic generated that would determine the rate of battery drain?

Lyr3x, 2022-01-03:
I would love to see real data. As soon as my dev PCB arrives I will setup a sensor with PIR etc with a power bank and see how it goes.

electricfire, 2022-01-03:
Your last statement sounds optimistic, i am still waiting on the sensor delivery i ordered it on Aliexpress hopefully they will be compatible with your firmware :) , i will try and have a play around

dmitriy-brynza, 2022-01-03:
> 

It is also better to disable the WiFi module and use the ZigBee module instead. It has a much lower power consumption, but requires a ZigBee hub.

Lyr3x, 2022-01-03:
Not a solution because we using ESPHome and that will not change. The goal is not to provide a battery powered first module.

mgernhard, 2022-01-08:
There will be coming matter. You can run this via "thread" its a 2,4GHz Bluetooth protocol which is using ipv6.
Apple, amazon and google will come up this year with more information Phillips hue will update as well. Maybe we get this for espHome as well.
Would solve some problems.

Lyr3x, 2022-01-08:
Well it will take a lot of time until matter become something real in the market. There is none yet and yes ESP32 (8266 can not) are technically matter compatible but it will take even more time until maybe ESPHome will support matter. I dont think this will happen sometime soon because its a complete different architecture. And matter will not really solve any battery related issues it still need to make the SoC sleep cause a continous connection with 2.4GHz draws battery.

closed 2022-05-29 by Lyr3x

Veuchez, 2022-11-27:
Good morning everyone,
Were you able to get results and run duration tests with battery power?
Thank you.