Lyr3x/Roode #84: How to use when the door width reaches 1.5m

sunshine-hass, 2022-01-03:
Hello, when the kitchen door is more than 1.5m away, where should the sensor be installed?
[image]

Lyr3x, 2022-01-03:
If possible still in the door frame. I will start to work on a two sensor solution soon. Then you may install two sensors over the door to catch the whole area

sunshine-hass, 2022-01-03:
[image]
[image]

Hello, is it as shown in the picture?
All three sensors will detect the action, and you need to use the program to determine the trigger signal action of the first sensor, and the other two sensors will not do the trigger processing.
There is air-conditioning above, the height is only 2.4m, and the floor height of the room is 2.8m

sunshine-hass, 2022-01-03:
Hello, I just sponsored this project. I hope to realize this function as soon as possible. I really need it!

Lyr3x, 2022-01-03:
With 2.4m height two sensors should be enough to count properly. It's definitely on the list, because more people are asking for it. No timeline yet as this needs quite some changes and testing.

sunshine-hass, 2022-01-03:
I have calculated the detection range of the sensor. The smaller the installation height, the smaller the range. If a person is not walking in the center, it will not be detected.
In fact, you can also write a script in Home Assistant to determine the 3 sensors, but I don't know how to write programs. .

Lyr3x, 2022-01-03:
I know how the sensor works :D 
 with an installation height auf 2.3m you can cover a 1.3m radius easily (i do this). And you dont need to walk in the center.

Tamershouman, 2022-10-22:
Hi There ,
My main target is to turn off the lights after my kids, and yes this is when the sensor is counting wrong.  With adults there is no issue at all, what would be the solution for that please? 

Thanks,

Lyr3x, 2022-10-22:
The only solution that comes to no mind would be placing the sensor horizontally in the door frame at a height were the kids could be pass easily. 
This is just a drawback of the sensor technology and I couldn't find any reliable solution till today. Small objects couldn't be traced reliable enough