Lyr3x/Roode #89: Return the people count as a "sensor" instead of "number"

Seba-VT, 2022-01-07:
**Is your feature request related to a problem? Please describe.**
Currently, the output as a `number` displays the sliding bar when using it on lovelace.

**Describe the solution you'd like**
I would like to have the result of the counting as a `sensor` in order to display its value without the sliding bar on lovelace. 

**Describe alternatives you've considered**
Tried to create a new sensor that capture the value of the output but its format is not compatible.
```
number:
  - platform: roode
    people_counter:
      id: count
      name: $friendly_name people counter

 sensor:
   - platform: template
     id: people_in_lab
     name: "People in Lab"
     lambda: !lambda |-
       return id(count);
```
I get an error during compilation saying that it can't convert from `PersistedNumber` to `float`

Lyr3x, 2022-01-07:
Why would you need a sensor ? We just refactored that cause a sensor does not make sense for a number. You not only have a slider but also access to the number. Unfortunately it's a float but that's cosmetic. You could add a separate entity to display a int if you really need.

Seba-VT, 2022-01-07:
Thanks for the quick replay @Lyr3x.

The issue that I have with the slider is that cover the number in lovelace when using it on my wall dashboard.
[image]

> You could add a separate entity to display a int if you really need.

How I can do that? I tried as I posted above but I get errors.

Lyr3x, 2022-01-07:
You have two options:
1.  https://github.com/thomasloven/lovelace-slider-entity-row
2. Create a template sensor:
```
 - platform: template
    name: "Template Sensor"
    lambda: |-
      return (int)id(peopleCounter).state;
```

Seba-VT, 2022-01-07:
> You have two options:
> 
> 1. https://github.com/thomasloven/lovelace-slider-entity-row
> 2. Create a template sensor:
> 
> ```
>  - platform: template
>     name: "Template Sensor"
>     lambda: |-
>       return (int)id(peopleCounter).state;
> ```

Thanks! I used the option 2 and now is correcly displayed. But now I have another issue, it takes several seconds to update the value to the template sensor, do you know why and how I can make it update in real time?

Lyr3x, 2022-01-07:
You coul add a on_value then update to the number sensor with something like. Don't have the exact syntax ready right now, but you can look it up for sure 👍 Otherwise, it's updated with its update interval.

Seba-VT, 2022-01-07:
I added `update_interval: 100ms` and now is working perfect!
Thank You for your help!

closed 2022-01-08 by Lyr3x