Lyr3x/Roode #91: Preserve last calibration results between reboots

gorbunovav, 2022-01-15:
**Is your feature request related to a problem? Please describe.**

The calibration process requires some stable and predictable conditions. 
The auto-calibration feature is useful since it allows to simplify the sensor setup.
Periodical re-calibration makes sense to adjust expectations to the changing environment.
Automatic re-calibration on boot is unpredictable.

**Describe the solution you'd like**

Preserve calibration results in some way. And don't perform automatic calibration on boot if data already present.
Re-calibration can be performed on schedule via the exposed service.

**Describe alternatives you've considered**

Use manual calibration (e.g. turn-on automatic method, grab values, fill into config). 
Downsides: 
 - each time sensor boots with some default values unrelated to the latest calibration results.
 - more manual work

CarsonF, 2022-01-15:
I'll work on this. It goes hand in hand with the sensors refactor I'm about to start.

Lyr3x, 2022-01-18:
@CarsonF This should be an optional configuration option.

CarsonF, 2022-01-18:
> This should be an optional configuration option.

As in not the default functionality?

Lyr3x, 2022-01-18:
I have #58 in mind and really want to implement that soon  because this eliminates the frequent recalibration issue. 

If #58 is done we this feature here can be `default: true`

gorbunovav, 2022-01-18:
@Lyr3x but this feature only prevents the re-calibration on boot. So in the described scenario disabling it will only help if you are rebooting the device regularly.
Although I agree that it can be configurable.

Lyr3x, 2022-01-18:
I think most of the users do reboot frequently to update the firmware and use the chance to get fresh thresholds in. I do understand why some don't want to do that always though 👍