Lyr3x/Roode #99: Manual ROI causes read distance failure

CarsonF, 2022-01-17:
**Describe the bug**
A clear and concise description of what the bug is.
All distance reads return 65535mm.

```yaml
roi: auto
```
Doesn't produce this behavior

**To Reproduce**
```yaml
# roi: 
```

CarsonF, 2022-01-17:
I believe I've found it. With default ROI center, reads are fine. After ROI calibration center is changed and reads fail

```
[14:54:24][I][Setup:212]: Calibrating sensor zones
[14:54:24][D][Zone:040]: ROI reset: { width: 6, height: 16, center: 167 }
[14:54:24][D][Zone:040]: ROI reset: { width: 6, height: 16, center: 231 }
[14:54:24][I][VL53L1X:155]: Set ranging mode: Medium
[14:54:24][D][Zone calibration:044]: Beginning. zoneId: 0
[14:54:24][V][VL53L1X:207]: Finished distance read: 991
[14:54:24][V][VL53L1X:207]: Finished distance read: 1140
[14:54:24][V][VL53L1X:207]: Finished distance read: 1258
[14:54:24][V][VL53L1X:207]: Finished distance read: 1158
[14:54:24][V][VL53L1X:207]: Finished distance read: 1065
[14:54:24][V][VL53L1X:207]: Finished distance read: 1089
[14:54:24][V][VL53L1X:207]: Finished distance read: 1193
[14:54:24][V][VL53L1X:207]: Finished distance read: 1087
[14:54:24][V][VL53L1X:207]: Finished distance read: 1167
[14:54:24][V][VL53L1X:207]: Finished distance read: 1366
[14:54:25][V][VL53L1X:207]: Finished distance read: 1072
[14:54:25][V][VL53L1X:207]: Finished distance read: 1147
[14:54:25][V][VL53L1X:207]: Finished distance read: 1270
[14:54:25][V][VL53L1X:207]: Finished distance read: 1168
[14:54:25][V][VL53L1X:207]: Finished distance read: 1257
[14:54:25][V][VL53L1X:207]: Finished distance read: 1128
[14:54:25][V][VL53L1X:207]: Finished distance read: 1149
[14:54:25][V][VL53L1X:207]: Finished distance read: 1070
[14:54:25][V][VL53L1X:207]: Finished distance read: 1233
[14:54:25][V][VL53L1X:207]: Finished distance read: 1242
[14:54:25][D][Zone calibration:123]: Zone AVG: 1162
[14:54:25][D][Zone calibration:124]: Zone SD: 93
[14:54:25][I][Zone calibration:063]: Calibrated threshold for zone. zoneId: 0, idle: 1069, min: 0 (0%), max: 908 (85%)
[14:54:25][D][Zone calibration:044]: Beginning. zoneId: 1
[14:54:25][V][VL53L1X:207]: Finished distance read: 1039
[14:54:25][V][VL53L1X:207]: Finished distance read: 1029
[14:54:25][V][VL53L1X:207]: Finished distance read: 1052
[14:54:25][V][VL53L1X:207]: Finished distance read: 1045
[14:54:25][V][VL53L1X:207]: Finished distance read: 1048
[14:54:25][V][VL53L1X:207]: Finished distance read: 1055
[14:54:25][V][VL53L1X:207]: Finished distance read: 1050
[14:54:25][V][VL53L1X:207]: Finished distance read: 1063
[14:54:25][V][VL53L1X:207]: Finished distance read: 1042
[14:54:25][V][VL53L1X:207]: Finished distance read: 1038
[14:54:25][V][VL53L1X:207]: Finished distance read: 1035
[14:54:25][V][VL53L1X:207]: Finished distance read: 1044
[14:54:25][V][VL53L1X:207]: Finished distance read: 1055
[14:54:25][V][VL53L1X:207]: Finished distance read: 1052
[14:54:25][V][VL53L1X:207]: Finished distance read: 1051
[14:54:25][V][VL53L1X:207]: Finished distance read: 1083
[14:54:25][V][VL53L1X:207]: Finished distance read: 1040
[14:54:25][V][VL53L1X:207]: Finished distance read: 1051
[14:54:25][V][VL53L1X:207]: Finished distance read: 1051
[14:54:25][V][VL53L1X:207]: Finished distance read: 1041
[14:54:25][D][Zone calibration:123]: Zone AVG: 1048
[14:54:25][D][Zone calibration:124]: Zone SD: 23
[14:54:25][I][Zone calibration:063]: Calibrated threshold for zone. zoneId: 1, idle: 1025, min: 0 (0%), max: 871 (85%)
[14:54:25][I][Zone calibration:108]: Calibrated ROI for zone. zoneId: 0, width: 6, height: 16, center: 150
[14:54:25][D][Zone calibration:044]: Beginning. zoneId: 0
[14:54:25][V][VL53L1X:207]: Finished distance read: 65535
[14:54:25][V][VL53L1X:207]: Finished distance read: 65535
[14:54:25][V][VL53L1X:207]: Finished distance read: 65535
[14:54:25][V][VL53L1X:207]: Finished distance read: 65535
[14:54:25][V][VL53L1X:207]: Finished distance read: 65535
[14:54:25][V][VL53L1X:207]: Finished distance read: 65535
[14:54:25][V][VL53L1X:207]: Finished distance read: 65535
[14:54:25][V][VL53L1X:207]: Finished distance read: 65535
[14:54:25][V][VL53L1X:207]: Finished distance read: 65535
[14:54:25][V][VL53L1X:207]: Finished distance read: 65535
[14:54:25][V][VL53L1X:207]: Finished distance read: 65535
[14:54:25][V][VL53L1X:207]: Finished distance read: 65535
[14:54:25][V][VL53L1X:207]: Finished distance read: 65535
[14:54:25][V][VL53L1X:207]: Finished distance read: 65535
[14:54:25][V][VL53L1X:207]: Finished distance read: 65535
[14:54:25][V][VL53L1X:207]: Finished distance read: 65535
[14:54:26][V][VL53L1X:207]: Finished distance read: 65535
[14:54:26][V][VL53L1X:207]: Finished distance read: 65535
[14:54:26][V][VL53L1X:207]: Finished distance read: 65535
[14:54:26][V][VL53L1X:207]: Finished distance read: 65535
[14:54:26][D][Zone calibration:123]: Zone AVG: 65535
[14:54:26][D][Zone calibration:124]: Zone SD: 0
[14:54:26][I][Zone calibration:063]: Calibrated threshold for zone. zoneId: 0, idle: 65535, min: 0 (0%), max: 55704 (85%)
[14:54:26][I][Zone calibration:108]: Calibrated ROI for zone. zoneId: 1, width: 6, height: 16, center: 247
[14:54:26][D][Zone calibration:044]: Beginning. zoneId: 1
[14:54:26][V][VL53L1X:207]: Finished distance read: 65535
[14:54:26][V][VL53L1X:207]: Finished distance read: 65535
[14:54:26][V][VL53L1X:207]: Finished distance read: 65535
[14:54:26][V][VL53L1X:207]: Finished distance read: 65535
[14:54:26][V][VL53L1X:207]: Finished distance read: 65535
[14:54:26][V][VL53L1X:207]: Finished distance read: 65535
[14:54:26][V][VL53L1X:207]: Finished distance read: 65535
[14:54:26][V][VL53L1X:207]: Finished distance read: 65535
[14:54:26][V][VL53L1X:207]: Finished distance read: 65535
[14:54:26][V][VL53L1X:207]: Finished distance read: 65535
[14:54:26][V][VL53L1X:207]: Finished distance read: 65535
[14:54:26][V][VL53L1X:207]: Finished distance read: 65535
[14:54:26][V][VL53L1X:207]: Finished distance read: 65535
[14:54:26][V][VL53L1X:207]: Finished distance read: 65535
[14:54:26][V][VL53L1X:207]: Finished distance read: 65535
[14:54:26][V][VL53L1X:207]: Finished distance read: 65535
```

CarsonF, 2022-01-17:
Ok I think I found it for real. We were automatically setting a calculated ROI center, even if width/height are fixed (which they are by default). This center calculation is based on the calculation for width/height. So when the center is based on a width/height that isn't actually used the measurements fail.

Lyr3x, 2022-01-18:
Yes that's a changed behavior after the refactor and might only work if the distance fits so the size is not to small and chooses a center which is too far on the outside of the grid. Good catch

PR #100 (2022-01-18): Fix measurements with default ROI
Fixes #99 

Base ROI center on the actual width used, not the just the one that's calculated. If a fixed width is used, like from config or default value, and it doesn't match the calculated one, then the ROI center set would not work.
closed 2022-01-18 by Lyr3x