Lyr3x/Roode #109: yaml doesn't compile services correctly

Bark33p3r, 2022-01-18:
**Describe the bug**
I'm not able to compile the api with the actual settings.

**To Reproduce**
use following in the yaml:
api:
  password: !secret api_password
  reboot_timeout: 60min
  services:
    - service: recalibrate
      then:
        - lambda: "id(roode_platform)->recalibration();"
    - service: set_max_threshold
      variables:
        newThreshold: int
      then:
        - lambda: "id(roode_platform)->set_max_threshold_percentage(newThreshold);id(roode_platform)->recalibration();"
    - service: set_min_threshold
      variables:
        newThreshold: int
      then:
        - lambda: "id(roode_platform)->set_min_threshold_percentage(newThreshold);id(roode_platform)->recalibration();"

Error:
/config/esphome/peoplecountermg.yaml: In lambda function:
/config/esphome/peoplecountermg.yaml:48:23: error: 'class esphome::roode::Roode' has no member named 'set_max_threshold_percentage'
         - lambda: "id(roode_platform)->set_max_threshold_percentage(newThreshold);id(roode_platform)->recalibration();"
                       ^
/config/esphome/peoplecountermg.yaml: In lambda function:
/config/esphome/peoplecountermg.yaml:53:23: error: 'class esphome::roode::Roode' has no member named 'set_min_threshold_percentage'
         - lambda: "id(roode_platform)->set_min_threshold_percentage(newThreshold);id(roode_platform)->recalibration();"
                       ^
*** [/data/peoplecountermg/.pioenvs/peoplecountermg/src/main.cpp.o] Error 1
========================== [FAILED] Took 2.10 seconds ==========================

It's possible to compile if the lower to services are commented out.
Would be nice for releases to comment them out if they doesn't work.

Thanks

Lyr3x, 2022-01-18:
Just remove max und min threshold calls for now.

Lyr3x, 2022-01-18:
Here are working example lambdas. You can easily create any service by looking up the function you want to call:
min/max threshold
```
- service: set_min_threshold_percentage
      variables:
        newThreshold: int
      then:
      - lambda: "id(roode_platform)->entry->threshold->set_min_percentage(newThreshold);id(roode_platform)->recalibration();"
    - service: set_max_threshold_percentage
      variables:
        newThreshold: int
      then:
      - lambda: "id(roode_platform)->entry->threshold->set_max_percentage(newThreshold);id(roode_platform)->recalibration();"
```

closed 2022-01-18 by Lyr3x