Lyr3x/Roode #120: Platform not found: 'text_sensor.roode'

noxhirsch, 2022-02-17:
**Describe the bug**
There seem to be some changes in the text sensor in ESPHome v2022.2.0, that the platform 'text_sensor.roode' isn't found anymore. If I remove my text sensors, I can update the device.
I assume that it might have something to do with this PR: https://github.com/esphome/esphome/pull/3172

**To Reproduce**
Steps to reproduce the behavior:
1. Update ESPHome
2. Click update on a Roode device with text sensors

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
```
INFO Reading configuration /config/esphome/az-people-counter.yaml...
INFO Updating https://github.com/Lyr3x/Roode@1.5.1
ERROR Unable to load component roode.text_sensor:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 162, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/esphome/.esphome/external_components/cb683fcf/components/roode/text_sensor.py", line 25, in <module>
    cv.Optional(CONF_ICON, default="mdi:git"): text_sensor.icon,
AttributeError: module 'esphome.components.text_sensor' has no attribute 'icon'
ERROR Unable to load component roode.text_sensor:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 162, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/esphome/.esphome/external_components/cb683fcf/components/roode/text_sensor.py", line 25, in <module>
    cv.Optional(CONF_ICON, default="mdi:git"): text_sensor.icon,
AttributeError: module 'esphome.components.text_sensor' has no attribute 'icon'
Failed config

text_sensor.roode: [source /config/esphome/az-people-counter.yaml:95]
  
  Platform not found: 'text_sensor.roode'.
  platform: roode
  version: 
    name: AZ PeopleCounter version
text_sensor.roode: [source /config/esphome/az-people-counter.yaml:98]
  
  Platform not found: 'text_sensor.roode'.
  platform: roode
  entry_exit_event: 
    name: AZ PeopleCounter last direction
```

Lyr3x, 2022-02-17:
Thanks for the report. Currently very busy, but I try to get it fixed quickly.
You are right it's about the changed Schema.

noxhirsch, 2022-02-17:
Don't stress yourself out about it! Only the update is blocked - the sensors work wonderfully without it :)

nijez, 2022-02-19:
Hi.
I am using Roode for the first time, and now I have the same problem. My ESPHOME is 2022.2.3, and Roode is the latest version 1.5.1.
May I ask if I use the old version or have to wait for the Roode update?

pepe59, 2022-02-19:
In the meantime, remove the text sensors and wait for the update. It will work without them.

nijez, 2022-02-20:
> 

How to remove the text sensors

pepe59, 2022-02-20:
Simply delete them in the node configuration

noxhirsch, 2022-02-20:
@nijez Or just comment them out with `#` at the beginning of every line:
[image]

nijez, 2022-02-21:
> text_sensor.

Thanks for your answer, now I have another error. what do i need to do?

Failed config

sensor.uptime: [source /config/esphome/peopleCounter32.yaml:147]
  platform: uptime
  name: Uptime Sensor
  id: uptime_sensor
  update_interval: 60s
  internal: True
  on_raw_value:  [source /config/esphome/peopleCounter32.yaml:153]
    then:  [source /config/esphome/peopleCounter32.yaml:154]
      - [source /config/esphome/peopleCounter32.yaml:154]
        
        Unable to find action with the name 'text_sensor.template.publish'.
        text_sensor.template.publish:  [source /config/esphome/peopleCounter32.yaml:155]
          id: uptime_human
          state: !lambda |-
            int seconds = round(id(uptime_sensor).raw_state);
            int days = seconds / (24 * 3600);
            seconds = seconds % (24 * 3600);
            int hours = seconds / 3600;
            seconds = seconds % 3600;
            int minutes = seconds /  60;
            seconds = seconds % 60;
            return (
              (days ? String(days) + "d " : "") +
              (hours ? String(hours) + "h " : "") +
              (minutes ? String(minutes) + "m " : "") +
              (String(seconds) + "s")
            ).c_str();

noxhirsch, 2022-02-21:
@nijez You can either comment out / remove the "Uptime Sensor", or re-enable the "Uptime Human Readable" text sensor (since it uses the platform "template" instead of Roode).

nijez, 2022-02-21:
[image]
either comment out / remove the "Uptime Sensor",

Lyr3x, 2022-02-21:
Remove those lambdas

PR #122 (2022-03-20): #120: Update text sensor definition to match ESPHome changes
Fixes #120
closed 2022-03-20 by Lyr3x