Lyr3x/Roode #125: Fix compatibility with ESPHome 2022.5.0

gorbunovav, 2022-05-27:
**Describe the bug**
With the ESPHome 2022.5.0 release the `set` method of the `Number` component was removed.

The proper way to set the value is to use the `make_call` method:
https://esphome.io/components/number/index.html?highlight=make_call

**To Reproduce**
Update to ESPHome 2022.5.0 and try to compile the firmware.

**Expected behavior**
Compilation completes succesfully.

Lyr3x, 2022-05-29:
Thanks. Yes unfortunately they removed that function. I saw that you are working on a PR? Happy to merge if you are ready.

gorbunovav, 2022-05-29:
@Lyr3x Hi! Yes, the PR is ready to be merged.

Lyr3x, 2022-05-29:
Thank you very much for the contribution

Lyr3x, 2022-06-01:
Can you open the PR to upstream please? 🙏

PR #126 (2022-06-01): #125: Fix compatibility with ESPHome 2022.5.0 
Fixes #125
commit a41e2519 (2022-06-01): Merge pull request #126 from gorbunovav/125-esphome-2022.5.0-compatibility-fix

#125: Fix compatibility with ESPHome 2022.5.0
closed 2022-06-01 by Lyr3x