Lyr3x/Roode #129: The sensor is seriously affected by light in daytime

roc634, 2022-08-24:
**Describe the bug**
The sensor seems to be seriously affected by light in daytime. I install the sensor in the door frame and use it to control the light in my kitchen. It works like a charm at night. However, it's annoying in daytime, especially in the morning. The light is always on. It'll turn on soon even though I turn if off manually . 
Below is my floor plan. My living room is facing east. The sunlight will reach my living room is the morning. The area in front of the ketchen door is brighter when the sunlight comes in, but it's still 2-3 meters away from the kitchen door. 
[image]
It's seriously affected by the sunlight while works fine during other parts of the day. 
[image]
[image]
[image]
As you can see, the distance measured after sunrise is a mess.
[image]
I've been trying different ways to solve this. I tilted the sensor. I've also changed several different power adapter but the situation is the same. I'm not sure if it's the problem of this individual sensor. Your help will be greatly appreciated. Thanks!

**Additional context**
By the way, have you tried the AMG8833 IR Thermal Camera? It is an 8x8 array of IR thermal sensors which is probably suitable for people counting, althought it's more expensive.

Lyr3x, 2022-08-28:
That is a known limitation of the sensor. There is nothing we can do about it. There are multiple issues about that already. 
I suggest you look into mmWave sensors instead if you face light reflection issues

roc634, 2022-08-28:
> That is a known limitation of the sensor. There is nothing we can do about it. There are multiple issues about that already. I suggest you look into mmWave sensors instead if you face light reflection issues

I'm also using APDS-9960 RGB and Gesture Sensor as people counter, it seems reliable, but it's detecting range is less than 20cm. It will be perfect if its detecting range is the same as GY-53.
I'm not that optimistic about mmWave sensors since they're detecting all moving objects rather than just human being.

Lyr3x, 2022-09-03:
This is true. A combination of sensors is key here. I currently use heavily motion sensor + light sensor + mmWave sensor with a perfect result.

closed 2022-10-22 by Lyr3x

artur90226, 2025-01-10:
> This is true. A combination of sensors is key here. I currently use heavily motion sensor + light sensor + mmWave sensor with a perfect result.

Which motion sensor are you using?