Lyr3x/Roode #130: Random disconnects while Wifi connection stays active

EuleMitKeule, 2022-09-01:
**Describe the bug**
All of my ESPs randomly disconnect every few days. They are still connected to my router's Wifi, but are unreachable either by their IP or their .local domain. A power cycle fixes this problem.

**To Reproduce**
Let the sensors run for some days and see the disconnected state in ESPHome.

**Expected behavior**
The sensors should stay connected over long periods of time.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
[configuration.txt](https://github.com/Lyr3x/Roode/files/9472570/configuration.txt)

Lyr3x, 2022-09-01:
That can happen for various reasons. Do you already have a indication on what's happening ?

EuleMitKeule, 2022-09-03:
I believe this was caused by a bad solder connection.

closed 2022-09-03 by EuleMitKeule