Lyr3x/Roode #133: Negative number for the counter

t112013, 2023-01-18:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is.
More or less yes
**Describe the solution you'd like**
A clear and concise description of what you want to happen.
The number to be able to be negative
**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.
The issue is about having two doors, so when people are leaving throw one door it wont go under 0.
I had to make another counter in HA and an automation based on the last direction of the sensor
Exposing number with the In and another with out would be perfect. 
Another solution would be to de able to define a second door with another sensor within the network

driv3r, 2023-02-01:
I was trying the `Exit` state change + virtual number value helper in HASS for my staircase project - it didn't work, as last entry/exit state it doesn't change when 2 people go in one direction, it would detect one `Exit` only. I've just changed now to having 5 being a "zero". This way I do something like: Add people counter from sensor A to people counter from sensor B and subtract 10, which gives me 0 when no one is on stairs and a positive number when there are multiple.

Lyr3x, 2023-04-10:
You can build an automation in HA to do the subtraction automatically. I've tested this briefly in the past, but do not have that situation anymore. 
I thought about adding this as a feature directly, but imho it does not make much sense as you can build this fairly easy in HA.

closed 2023-04-10 by Lyr3x