Lyr3x/Roode #134: Not working from last update Home assistant and esphome

dmitijmaslak, 2023-02-05:
**Describe the bug**
```
`INFO Reading configuration /config/esphome/peoplecounteresphome.yaml...
INFO Updating https://github.com/Lyr3x/Roode@master
INFO Starting log output from 192.168.1.42 using esphome API
INFO Successfully connected to 192.168.1.42
[15:17:21][I][app:102]: ESPHome version 2022.12.8 compiled on Feb  5 2023, 15:09:21
[15:17:21][I][i2c.arduino:068]: Results from i2c bus scan:
[15:17:21][E][i2c.arduino:076]: Unknown error at address 0x08
[15:17:21][E][i2c.arduino:076]: Unknown error at address 0x09
[15:17:21][E][i2c.arduino:076]: Unknown error at address 0x0A
[15:17:21][E][i2c.arduino:076]: Unknown error at address 0x0B
[15:17:21][E][i2c.arduino:076]: Unknown error at address 0x0C
[15:17:21][E][i2c.arduino:076]: Unknown error at address 0x0D
[15:17:21][E][i2c.arduino:076]: Unknown error at address 0x0E
[15:17:21][E][i2c.arduino:076]: Unknown error at address 0x0F
[15:17:21][E][i2c.arduino:076]: Unknown error at address 0x10
[15:17:21][E][i2c.arduino:076]: Unknown error at address 0x11
[15:17:21][E][i2c.arduino:076]: Unknown error at address 0x12
[15:17:21][E][i2c.arduino:076]: Unknown error at address 0x13
[15:17:21][E][i2c.arduino:076]: Unknown error at address 0x14
[15:17:21][E][i2c.arduino:076]: Unknown error at address 0x15
[15:17:21][E][i2c.arduino:076]: Unknown error at address 0x16
[15:17:21][E][i2c.arduino:076]: Unknown error at address 0x17
[15:17:21][E][i2c.arduino:076]: Unknown error at address 0x18
[15:17:21][E][i2c.arduino:076]: Unknown error at address 0x19
[15:17:21][E][i2c.arduino:076]: Unknown error at address 0x1A
[15:17:21][E][i2c.arduino:076]: Unknown error at address 0x1B
[15:17:21][E][i2c.arduino:076]: Unknown error at address 0x1C
[15:17:21][E][i2c.arduino:076]: Unknown error at address 0x1D
[15:17:21][E][i2c.arduino:076]: Unknown error at address 0x1E
[15:17:21][E][i2c.arduino:076]: Unknown error at address 0x1F
[15:17:21][E][i2c.arduino:076]: Unknown error at address 0x20
[15:17:21][E][i2c.arduino:076]: Unknown error at address 0x21
[15:17:21][E][i2c.arduino:076]: Unknown error at address 0x22
[15:17:21][E][i2c.arduino:076]: Unknown error at address 0x23
[15:17:21][E][i2c.arduino:076]: Unknown error at address 0x24
[15:17:21][E][i2c.arduino:076]: Unknown error at address 0x25
[15:17:22][E][i2c.arduino:076]: Unknown error at address 0x26
[15:17:22][E][i2c.arduino:076]: Unknown error at address 0x27
[15:17:22][E][i2c.arduino:076]: Unknown error at address 0x28
[15:17:22][E][i2c.arduino:076]: Unknown error at address 0x29
[15:17:22][E][i2c.arduino:076]: Unknown error at address 0x2A
[15:17:22][E][i2c.arduino:076]: Unknown error at address 0x2B
[15:17:22][E][i2c.arduino:076]: Unknown error at address 0x2C
[15:17:22][E][i2c.arduino:076]: Unknown error at address 0x2D
[15:17:22][E][i2c.arduino:076]: Unknown error at address 0x2E
[15:17:22][E][i2c.arduino:076]: Unknown error at address 0x2F
[15:17:22][E][i2c.arduino:076]: Unknown error at address 0x30
[15:17:22][E][i2c.arduino:076]: Unknown error at address 0x31
[15:17:22][E][i2c.arduino:076]: Unknown error at address 0x32
[15:17:22][E][i2c.arduino:076]: Unknown error at address 0x33
[15:17:22][E][i2c.arduino:076]: Unknown error at address 0x34
[15:17:22][E][i2c.arduino:076]: Unknown error at address 0x35
[15:17:22][E][i2c.arduino:076]: Unknown error at address 0x36
[15:17:22][E][i2c.arduino:076]: Unknown error at address 0x37
[15:17:22][E][i2c.arduino:076]: Unknown error at address 0x38
[15:17:22][E][i2c.arduino:076]: Unknown error at address 0x39
[15:17:22][E][i2c.arduino:076]: Unknown error at address 0x3A
[15:17:22][E][i2c.arduino:076]: Unknown error at address 0x3B
[15:17:22][E][i2c.arduino:076]: Unknown error at address 0x3C
[15:17:22][E][i2c.arduino:076]: Unknown error at address 0x3D
[15:17:22][E][i2c.arduino:076]: Unknown error at address 0x3E
[15:17:22][E][i2c.arduino:076]: Unknown error at address 0x3F
[15:17:22][E][i2c.arduino:076]: Unknown error at address 0x40
[15:17:22][E][i2c.arduino:076]: Unknown error at address 0x41
[15:17:22][E][i2c.arduino:076]: Unknown error at address 0x42
[15:17:22][E][i2c.arduino:076]: Unknown error at address 0x43
[15:17:22][E][i2c.arduino:076]: Unknown error at address 0x44
[15:17:22][E][i2c.arduino:076]: Unknown error at address 0x45
[... 212 lines cut ...]
              seconds = seconds % (24 * 3600);
              int hours = seconds / 3600;
              seconds = seconds % 3600;
              int minutes = seconds /  60;
              seconds = seconds % 60;
              return (
                (days ? String(days) + "d " : "") +
                (hours ? String(hours) + "h " : "") +
                (minutes ? String(minutes) + "m " : "") +
                (String(seconds) + "s")
              ).c_str();

text_sensor:
  - platform: roode
    version:
      name: $friendly_name version
  - platform: roode
    entry_exit_event:
      name: $friendly_name last direction
  - platform: template
    name: $friendly_name Uptime Human Readable
    id: uptime_human
    icon: mdi:clock-start

```

Lyr3x, 2023-02-05:
Currently not, but I have no time to look into. If you find out you can open a PR.

dmitijmaslak, 2023-02-07:
Sorry my bad...
bad contact....

closed 2023-02-07 by dmitijmaslak